ArduPilot/ardupilot_gazebo #25: gazebo server crash on macOS when gazebo gui launch

xlla, 2022-05-29:
## Bug report

**Issue details**
gazebo server will crash at once on macOS when gazebo gui launch

**Version**
ignition-garden branch

**Logs**

server launch first
<pre>
ign gazebo -s -v 4 -r iris_arducopter_runway.world
[Msg] Ignition Gazebo Server v7.0.0~pre1
this->dataPtr->config.Source() 2
[Msg] Loading default world.
[Msg] Loading SDF world file[/Users/xlla/git/ardupilot_gazebo-1/worlds/iris_arducopter_runway.world].
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Dbg] [ServerConfig.cc:1004] Loaded (3) plugins from file [/Users/xlla/.ignition/gazebo/7/server.config]
[Dbg] [Physics.cc:824] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/local/Cellar/ignition-physics6/5.999.999~0~20220414/lib/ign-physics-6/engine-plugins/libignition-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:89] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Msg] Create service on [/world/iris_arducopter_runway/create]
[Msg] Remove service on [/world/iris_arducopter_runway/remove]
[Msg] Pose service on [/world/iris_arducopter_runway/set_pose]
[Msg] Pose service on [/world/iris_arducopter_runway/set_pose_vector]
[Msg] Light configuration service on [/world/iris_arducopter_runway/light_config]
[Msg] Physics service on [/world/iris_arducopter_runway/set_physics]
[Msg] SphericalCoordinates service on [/world/iris_arducopter_runway/set_spherical_coordinates]
[Msg] Enable collision service on [/world/iris_arducopter_runway/enable_collision]
[Msg] Disable collision service on [/world/iris_arducopter_runway/disable_collision]
[Msg] Material service on [/world/iris_arducopter_runway/visual_config]
[Msg] Material service on [/world/iris_arducopter_runway/wheel_slip]
[Dbg] [SystemManager.cc:89] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:89] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Msg] Serving world controls on [/world/iris_arducopter_runway/control], [/world/iris_arducopter_runway/control/state] and [/world/iris_arducopter_runway/playback/control]
[Msg] Serving GUI information on [/world/iris_arducopter_runway/gui/info]
[Msg] World [iris_arducopter_runway] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/iris_arducopter_runway/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Dbg] [SystemManager.cc:89] Loaded system [ignition::gazebo::systems::JointStatePublisher] for entity [9]
[Dbg] [SystemManager.cc:89] Loaded system [ignition::gazebo::systems::LiftDrag] for entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::LiftDrag] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::LiftDrag] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::LiftDrag] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::LiftDrag] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::LiftDrag] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::LiftDrag] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::LiftDrag] is already loaded in this entity [9]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris/joint/rotor_0_joint/cmd_force]
[Dbg] [SystemManager.cc:89] Loaded system [ignition::gazebo::systems::ApplyJointForce] for entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::ApplyJointForce] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::ApplyJointForce] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:61] This system Plugin [ignition::gazebo::systems::ApplyJointForce] is already loaded in this entity [9]
[Dbg] [SystemManager.cc:89] Loaded system [ArduPilotPlugin] for entity [9]
[Dbg] [Physics.cc:824] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/local/Cellar/ignition-physics6/5.999.999~0~20220414/lib/ign-physics-6/engine-plugins/libignition-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:89] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Msg] Create service on [/world/iris_arducopter_runway/create]
[Msg] Remove service on [/world/iris_arducopter_runway/remove]
[Msg] Pose service on [/world/iris_arducopter_runway/set_pose]
[Msg] Pose service on [/world/iris_arducopter_runway/set_pose_vector]
[Msg] Light configuration service on [/world/iris_arducopter_runway/light_config]
[Msg] Physics service on [/world/iris_arducopter_runway/set_physics]
[Msg] SphericalCoordinates service on [/world/iris_arducopter_runway/set_spherical_coordinates]
[Msg] Enable collision service on [/world/iris_arducopter_runway/enable_collision]
[Msg] Disable collision service on [/world/iris_arducopter_runway/disable_collision]
[... 152 lines cut ...]
[GUI] [Msg] Follow service on [/gui/follow]
[GUI] [Msg] Move to pose service on [/gui/move_to/pose]
[GUI] [Msg] Camera pose topic advertised on [/gui/camera/pose]
[GUI] [Msg] Follow offset service on [/gui/follow/offset]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [InteractiveViewControl.cc:130] InteractiveViewControl plugin is moving camera [scene::Camera(65527)]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]

</pre>

server will crash 
<pre>
Stack trace (most recent call last) in thread 123145340256256:
[Dbg] [EntityComponentManager.cc:1617] Updated state thread iterators: 8 threads processing around 5 entities each.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v135WorldE], which doesn't have `operator<<`. Component will not be serialized.
#4    Object "libsystem_pthread.dylib", at 0x7fff55b2850c, in _pthread_start + 376
#3    Object "libsystem_pthread.dylib", at 0x7fff55b28660, in _pthread_body + 339
[Wrn] [Component.hh:144] Trying to serialize component with data type [#2    Object "libignition-gazebo7.7.dylib", at NSt3__16vectorIdNS_9allocatorIdEEEE0x10add2292, in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, ignition::gazebo::v7::SimulationRunner::ProcessSystemQueue()::$_0> >(void*) + 402
], which doesn't have #1    Object "libignition-gazebo-scene-broadcaster-system.dylib", at `operator<<`. Component will not be serialized.0x11b46beb2, in ignition::gazebo::v7::systems::SceneBroadcaster::PostUpdate(ignition::gazebo::v7::UpdateInfo const&, ignition::gazebo::v7::EntityComponentManager const&) + 1618

#0    Object "libignition-gazebo7.7.dylib", at 0x10ad19281, in ignition::gazebo::v7::EntityComponentManager::State(ignition::msgs::SerializedStateMap&, std::__1::unordered_set<unsigned long long, std::__1::hash<unsigned long long>, std::__1::equal_to<unsigned long long>, std::__1::allocator<unsigned long long> > const&, std::__1::unordered_set<unsigned long long, std::__1::hash<unsigned long long>, std::__1::equal_to<unsigned long long>, std::__1::allocator<unsigned long long> > const&, bool) const + 273
Segmentation fault: 11

</pre>

xlla, 2022-05-29:
I think the server side crash log is fuzzy due to multi-thread output on same terminal, it is not cause by github's html render.
[image]

srmainwaring, 2022-05-29:
Hi @xlla, I've tried to replicate your issue running the ignition-garden branch of ardupilot_gazebo against the most recent dev versions of Gazebo Garden. AFAICT everything is working as expected.

There is one update required (to the ignition-cmake dependency - which is now version 3) when using the current Garden version. It could be that you have some stale dependencies or a library conflict somewhere (depending on whether you have tried to install other OSRF libraries using homebrew)?

xlla, 2022-05-31:
Hi @srmainwaring, I found the reason.

I have built all depend libraries base on main branch except ogre-next on branch v2-2, using same cmake args copied from brew installation logs.

for ardupilot_gazebo, at first, I use a fork of your fork base on branch ignition-garden, it's latest commit was Fri Nov 5.
until read your reply, I switch to the ignition-garden branch of ardupilot_gazebo, it's latest commit was Thu Mar 31. 
but the issue remain.

then I have add many igndbg trace statements, I found EntityComponentManager::State maybe have concurrency problem.

I remember now, in recently, I have merged  [gz-sim PR1484](https://github.com/gazebosim/gz-sim/pull/1484), after revert this PR,  gazebo works again!

We can close issue, thanks!