ArduPilot/ardupilot_gazebo #33: Add fetature/plugin for parachute testing

41Mo, 2022-12-15:
# Feature request

**It would be nice to be able to test parachute release with gazebo.**

**We could make something similar**
px4 gazebo plugins already have this feature, but for gazebo-classic

**I have made my own implementation, but its poorly written**
[ParachutePlugin.cc](https://github.com/41Mo/ardupilot_gazebo/blob/parachute/src/ParachutePlugin.cc)

**Platform**
[  ] All
[  ] AntennaTracker
[ ✓ ] Copter
[ ✓ ] Plane
[  ] Rover
[  ] Submarine

srmainwaring, 2022-12-16:
@41Mo, if you rebase your topic branch https://github.com/41Mo/ardupilot_gazebo/tree/parachute on `main` I'd be happy to work on this with you. It would be great to expand the range of ArduPilot features available in Gazebo.

Not a hard requirement, but if you could name your topic branches:

`<username>/topic` (eg. 41Mo/parachute)

that helps keep things organised on my side when I pull your branch in for review.

srmainwaring, 2022-12-16:
Initial thoughts:

1. ArduPilotPlugin.cc

We can reuse the existing COMMAND control type for the parachute deployment. This was added to forward PWM levels onto the thruster plugin for a ROV, but I think we can use the same approach here rather than add a new command type and additional fields to the `<control>` element. The parachute plugin would then decide how to interpret the forwarded PWM level (i.e. high for deploy or whatever).

The following would send the output of `SERVO5` onto `zephyr/parachute/deploy`:

```xml
<plugin name="ArduPilotPlugin" filename="libArduPilotPlugin">
  ...   
  <control channel="4">
    ...
    <type>COMMAND</type>
    <cmd_topic>zephyr/parachute/deploy</cmd_topic>
    ...
  </control>
</plugin>
```

2. ParachutePlugin.cc

It may not scale well having this plugin attached to the world. The parachute is attached to a model, so the plugin should be too. Still working through the implementation details so will update this section in due course...

41Mo, 2022-12-16:
I completely agree. 
I originally created this as a POC, and my knowledge with gazebo plugins is not good enough.
When you will be ready with your thoughts i can create a draft pr.

srmainwaring, 2022-12-16:
It looks like a good start - I don't think the COMMAND type was there when you developed the plugin.

I'll experiment with what you've done to see if I can get the plugin localised to a model. Would you mind jotting down some instructions how to run an example - I'm not familiar with the parachute deployment function in ArduPilot. A list of terminal and mavproxy commands would be a great help.

srmainwaring, 2022-12-17:
@41Mo, I have a slightly modified version of your plugin running - it's very nice! Great work with the detachable joints.

I've made a few cosmetic changes - mainly moving implementation details into a private class in the C++ file and some formatting details to keep cpplint happy.

Next steps:

- [ ] Make the changes to use the COMMAND keyword so are no changes are required to `ArduPilotPlugin.cc` and all state logic is managed in `ParachutePlugin.cc`.
- [ ] Make `ParachutePlugin` a model plugin attached to the Zephyr rather than the world.
- [ ] SDF param name changes - we can probably refer to the vehicle link and parachute link concretely rather than in the abstract as parent and child.
- [ ] Try using a universal joint (or equiv) rather than fixed. This should let the vehicle pivot under the parachute more naturally.
- [ ] General review of the state transitions and document some tests to verify it's all working. 

https://user-images.githubusercontent.com/24916364/208252069-ec8898f8-65d9-4cbc-9ff3-a4a28fd7d907.mov

41Mo, 2022-12-17:
Glad to hear it.
Good job!

srmainwaring, 2022-12-19:
@41Mo - posted a PR building on your initial version. Would appreciate your feedback and testing if you get a chance. It's stripped down to minimal functionality. Can add the detach commands back in if required (but I don't think ArduPilot supports that - so I left it out of this version).

41Mo, 2022-12-19:
@srmainwaring 
Perfect. 
I will try to give feedback this week.

Detach is not required.
I have implemented this function to test decouple feature of my aircraft.

srmainwaring, 2022-12-19:
Thanks. The main issue I found was that for certain choices of the release pose the collision engine would crash (hard assert and exit signal). I think this was because the original inertial for the parachute resulted in an instability arising from the impulse generated if the chute and vehicle collided before the joint was created - setting a pose directly rather than indirectly through applied forces can sometimes cause these instabilities. I reworked the collision (it is now a sphere with diameter of the canopy) and recalculated the inertial and set the CoM at the collision origin. This seems to have fixed the issue, but would be interested to know if it breaks for some inputs choices.

PR #36 (2022-12-22): Add parachute plugin
This PR adds a parachute plugin requested in #33 and builds on the initial version by @41Mo.

- Closes #33.

## Details

- Add a system plugin `ParachutePlugin` that can be included in a `<model>...</model>` element.
- Add a version of the zephyr model with the `ArduPilotPlugin` configured to forward servo commands for parachute release to the `gz topic` `parachute/command`.
- Add an example world `zephyr_parachute.sdf`.
- Add a test world `test_parachute.sdf` that adds a parachute to a box that can be deployed using the `gz topic` command line to publish a release command [... cut]
closed 2022-12-22 by srmainwaring