ArduPilot/ardupilot_gazebo #37: Add Gimbal

srmainwaring, 2022-12-21:
## Feature request

The original version of the project for Gazebo11 included a plugin for a small 2d gimbal. This request is to have the gimbal model updated to run in Gazebo Sim and for the plugin to be ported to work with the `gz-sim7` libraries.  

#### Desired outcome

- A small gimbal model is available to attach to the Iris quadcopter.
- The gimbal joints may be controlled using a suitable plugin.
- There is a  topic to which commands may be published to set the gimbal joint position.

#### Alternatives

The gimbal model is available for download from [Gimbal Small 2D](https://app.gazebosim.org/OpenRobotics/fuel/models/Gimbal%20Small%202D) but has not been ported to Gazebo Sim / PBS shading:

- It uses the deprecated material `<script>` element for specifying visual colours.
- It uses the plugin `libGimbalSmall2dPlugin` which does not work in Gazebo Sim.

#### Platform

[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[ x ] Rover
[ x ] Submarine

#### Additional context

- ArduPilot Discourse: [Gimbal does not move in Gazebo](https://discuss.ardupilot.org/t/gimbal-does-not-move-in-gazebo/36859)

srmainwaring, 2022-12-21:
Tracking development on https://github.com/srmainwaring/ardupilot_gazebo-1/tree/srmainwaring/garden/gimbal-plugin

- [ ] Update `gimbal_small_2d model` to render in Gazebo Sim.
- [ ] Add an arm joint in `gimbal_small_2d` for an extra DOF.
- [ ] Update `iris_with_ardupilot` model to enable gimbal.
- [ ] Add plugin for joint position control (reuse `gz::sim::systems::JointPositionControl` if it works for nested models). 
- [ ] Add example world and configure the `ArduPilotPlugin` to forward servo commands to the joint position control topics.

### Progress 

Model, additional joint and camera sensor rendering correctly in Gazebo.

[image]

closed 2023-02-09 by srmainwaring

chngdickson, 2023-03-31:
@srmainwaring How do we control said gimbal using ardupilot's plugin?

srmainwaring, 2023-03-31:
> How do we control said gimbal using ardupilot's plugin?

@chngdickson, the PR https://github.com/ArduPilot/ardupilot_gazebo/pull/43 includes a note on how to control the gimbal from MAVProxy using RC overrides. The example is set up as a standard servo gimbal so you should be able to follow the wiki guide https://ardupilot.org/copter/docs/common-camera-gimbal.html with the Gazebo version.

samiamlabs, 2024-01-04:
Hi @srmainwaring!

I have been trying to control the camera tilt on a simulated bluerov2 using ardusub and the camera tilt RC input (https://www.ardusub.com/developers/rc-input-and-output.html).

I used this and #43 as a reference. Controlling it using the gz topic works now, but I seem to be getting only a static value on control channel 16 (`<control channel="15">`) from the ardupilot gazebo plugin. This is the output channel used for tilt on the physical bluerov2 https://bluerobotics.com/store/comm-control-power/control/navigator/.

Do you happen to know if it should be possible to control a simulated camera gimbal this way using ardusub? I'm guessing I may need to add something to my .param file, but am not sure what, or where to find more info about it.

srmainwaring, 2024-01-04:
Hi @samiamlabs,

We should be able to sort something out. This issue is closed now, so it would be better to follow up on ArduPilot Discourse (https://discuss.ardupilot.org/). If you add a topic that includes `gazebo`, `gimbal` and tag me we can pick it up there. May also be good to add the BlueROV2 model to ArduPilot/SITL_Models so we can provide a standard configuration with a few variants for common payloads.

PR #43 (2023-02-09): Gimbal: enable gimbal in iris model
This PR enables a gimbal and camera sensor in the `iris_with_gimbal` model. The gimbal has 2 DOF and uses the Gazebo `JointPositionController` rather than a custom plugin.

Closes #37

## Details

1. Update gimbal models
- Update joint limits.
- Rename arm_link, arm_joint to roll_link, roll_joint.
- Remove blank lines.

2. Update gimbal in iris
- Add `<name>` element when including the gimbal to alias the model name to `gimbal`.
- Add control channels for roll and tilt.
- Enable the joint position controllers to actuate the gimbal joints.

3. Remove deprecat [... cut]