ArduPilot/ardupilot_gazebo #39: Support for gz-fortress (LTS)

julled, 2022-12-21:
## Feature request

**Is your feature request related to a problem? Please describe.**
So far i am bound to ROS1/Noetic and i want to use gazebo image bridge from https://github.com/gazebosim/ros_gz/tree/noetic. The ROS noetic codebase over there only supports the LTS release so far. This leeds to the problem, that AFAIK no ROS1 codebase can be used with [ardupilot_gazebo](https://github.com/ArduPilot/ardupilot_gazebo).

**Describe the solution you'd like**
Add support for gz-fortress (LTS).

**Describe alternatives you've considered**
If using ROS2 would be a option, then https://github.com/gazebosim/ros_gz is possible. For ROS1 there is AFAIK no other option avaiable which works with this repo.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

julled, 2022-12-21:
I just realized there is a branch for fortress: https://github.com/ArduPilot/ardupilot_gazebo/tree/fortress

I tried to get the zephyr world running using the SITL start procedure from the main branch, but without success.

Do you have some example startup procedures which work for the old fortress branch?

srmainwaring, 2022-12-21:
@julled - a couple of questions:

1. Is the Iris quad-copter model working?
2. Is the zephyr responding to `rc` commands, for instance:

  ```bash
  # arm
  MANUAL> arm throttle
  # spin prop
  MANUAL> rc 3 1200
  # both elevons up / dn
  MANUAL> rc 2 1100
  MANUAL> rc 2 1900
  # elevons in opposite directions
  MANUAL> rc 1 1100
  MANUAL> rc 1 1900
  ```
If yes to the above, then the lack of controlled flight may be due to the servo parameters being reversed in the plugin compared to the SITL parameters for gazebo-zephyr. Try applying the change in https://github.com/ArduPilot/ardupilot_gazebo/pull/24 which was not back-ported to the fortress branch.

srmainwaring, 2023-05-19:
Closing as no recent activity and a `fortress` branch is available (although not actively updated).

closed 2023-05-19 by srmainwaring