ArduPilot/ardupilot_gazebo #45: Error: [ruby $(which gz) sim-1] ODE INTERNAL ERROR 1: assertion "aabbBound >= dMinIntExact && aabbBound < dMaxIntExact" failed in collide() [collision_space.cpp:460]

Rezenders, 2023-02-06:
## Bug report

When I run a gazebo simulation with the ardupilot plugin loaded, it crashes immediately when ardupilot connects via mavros. This started to happen after gazebo got upgraded to version 7.2. More info on: https://github.com/gazebosim/gz-sim/issues/1871

I am not sure if this is a bug related to the plugin or gazebo.

gazebo output:

```
ros2 launch bluerov2_ignition simulation.launch.py 
[INFO] [launch]: All log files can be found below /home/gus/.ros/log/2023-01-31-13-37-20-944372-alien-362949
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which gz) sim-1]: process started with pid [362952]
[INFO] [create-2]: process started with pid [362954]
[INFO] [create-3]: process started with pid [362957]
[INFO] [create-4]: process started with pid [362959]
[INFO] [parameter_bridge-5]: process started with pid [362962]
[parameter_bridge-5] [INFO] [1675168641.110988456] [gz_bluerov_pose_bridge]: Creating GZ->ROS Bridge: [/model/bluerov2/pose (gz.msgs.Pose) -> /model/bluerov2/pose (geometry_msgs/msg/Pose)] (Lazy 0)
[parameter_bridge-5] [INFO] [1675168641.113575214] [gz_bluerov_pose_bridge]: Creating ROS->GZ Bridge: [/model/bluerov2/pose (geometry_msgs/msg/Pose) -> /model/bluerov2/pose (gz.msgs.Pose)] (Lazy 0)
[create-2] [INFO] [1675168641.586280364] [ros_gz_sim]: Requested creation of entity.
[create-2] [INFO] [1675168641.586449620] [ros_gz_sim]: OK creation of entity.
[create-3] [INFO] [1675168641.686370576] [ros_gz_sim]: Requested creation of entity.
[create-3] [INFO] [1675168641.686529413] [ros_gz_sim]: OK creation of entity.
[INFO] [create-2]: process has finished cleanly [pid 362954]
[create-4] [INFO] [1675168641.786736353] [ros_gz_sim]: Requested creation of entity.
[create-4] [INFO] [1675168641.787018690] [ros_gz_sim]: OK creation of entity.
[INFO] [create-3]: process has finished cleanly [pid 362957]
[INFO] [create-4]: process has finished cleanly [pid 362959]
[ruby $(which gz) sim-1] [Err] [SDFFeatures.cc:819] The geometry element of collision [ground] couldn't be created
[ruby $(which gz) sim-1] 
[ruby $(which gz) sim-1] ODE INTERNAL ERROR 1: assertion "aabbBound >= dMinIntExact && aabbBound < dMaxIntExact" failed in collide() [collision_space.cpp:460]
[ruby $(which gz) sim-1] Stack trace (most recent call last):
[ruby $(which gz) sim-1] #31   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f4935642c96, in 
[ruby $(which gz) sim-1] #30   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f493563ffc5, in 
[ruby $(which gz) sim-1] #29   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f493563dc34, in 
[ruby $(which gz) sim-1] #28   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f4935589a1e, in 
[ruby $(which gz) sim-1] #27   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f49354b49ac, in rb_protect
[ruby $(which gz) sim-1] #26   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f493564cc61, in rb_yield
[ruby $(which gz) sim-1] #25   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f493564830c, in rb_vm_exec
[ruby $(which gz) sim-1] #24   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f4935642c96, in 
[ruby $(which gz) sim-1] #23   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f493563ffc5, in 
[ruby $(which gz) sim-1] #22   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f493563dc34, in 
[ruby $(which gz) sim-1] #21   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7f49314fb44b, in 
[ruby $(which gz) sim-1] #20   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f493560b088, in rb_nogvl
[ruby $(which gz) sim-1] #19   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7f49314fad6b, in 
[ruby $(which gz) sim-1] #18   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f49314ec492, in 
[ruby $(which gz) sim-1] #17   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f49314efe2d, in 
[ruby $(which gz) sim-1] #16   Object "/usr/lib/x86_64-linux-gnu/libgz-sim7-gz.so.7.2.0", at 0x7f4930a09e35, in runServer
[ruby $(which gz) sim-1] #15   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f49306e3f55, in 
[ruby $(which gz) sim-1] #14   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f49306f13aa, in gz::sim::v7::SimulationRunner::Run(unsigned long)
[ruby $(which gz) sim-1] #13   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f49306f0a80, in gz::sim::v7::SimulationRunner::Step(gz::sim::v7::UpdateInfo const&)
[ruby $(which gz) sim-1] #12   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f49306eefb2, in gz::sim::v7::SimulationRunner::UpdateSystems()
[ruby $(which gz) sim-1] #11   Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-physics-system.so", at 0x7f49207262e6, in gz::sim::v7::systems::Physics::Update(gz::sim::v7::UpdateInfo const&, gz::sim::v7::EntityComponentManager&)
[ruby $(which gz) sim-1] #10   Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-physics-system.so", at 0x7f4920726961, in gz::sim::v7::systems::PhysicsPrivate::Step(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&)
[ruby $(which gz) sim-1] #9    Object "/usr/lib/x86_64-linux-gnu/gz-physics-6/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f48fee6e28c, in gz::physics::dartsim::SimulationFeatures::WorldForwardStep(gz::physics::Identity const&, gz::physics::SpecifyData<gz::physics::RequireData<gz::physics::WorldPoses>, gz::physics::ExpectData<gz::physics::ChangedWorldPoses, gz::physics::Contacts, gz::physics::JointPositions> >&, gz::physics::CompositeData&, gz::physics::ExpectData<gz::physics::ApplyExternalForceTorques, gz::physics::ApplyGeneralizedForces, gz::physics::VelocityControlCommands, gz::physics::ServoControlCommands, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&)
[ruby $(which gz) sim-1] #8    Object "/lib/x86_64-linux-gnu/libdart.so.6.12", at 0x7f48feab6e73, in dart::simulation::World::step(bool)
[ruby $(which gz) sim-1] #7    Object "/lib/x86_64-linux-gnu/libdart.so.6.12", at 0x7f48fea9c705, in dart::constraint::ConstraintSolver::solve()
[ruby $(which gz) sim-1] #6    Object "/lib/x86_64-linux-gnu/libdart.so.6.12", at 0x7f48fea9afe1, in dart::constraint::ConstraintSolver::updateConstraints()
[ruby $(which gz) sim-1] #5    Object "/lib/x86_64-linux-gnu/libdart-collision-ode.so.6.12", at 0x7f48fec9e1cb, in dart::collision::OdeCollisionDetector::collide(dart::collision::CollisionGroup*, dart::collision::CollisionOption const&, dart::collision::CollisionResult*)
[ruby $(which gz) sim-1] #4    Object "/lib/x86_64-linux-gnu/libode.so.8", at 0x7f48fe599266, in dxHashSpace::collide(void*, void (*)(void*, dxGeom*, dxGeom*))
[ruby $(which gz) sim-1] #3    Object "/lib/x86_64-linux-gnu/libode.so.8", at 0x7f48fe5a17b7, in dDebug
[ruby $(which gz) sim-1] #2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f493520e7f2, in abort
[ruby $(which gz) sim-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4935228475, in raise
[ruby $(which gz) sim-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f493527ca7c, in pthread_kill
[ruby $(which gz) sim-1] Aborted (Signal sent by tkill() 363028 1000)
[ruby $(which gz) sim-1] libEGL warning: DRI2: failed to create dri screen
[ruby $(which gz) sim-1] libEGL warning: DRI2: failed to create dri screen
[INFO] [ruby $(which gz) sim-1]: process has finished cleanly [pid 362952]

```

**Version**
- Ubuntu 22.04
- Mavros 2.4.0
- ardupilot_gazebo main branch
- ros_gz humble branch
- gazebo 7.3.0

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ X ] Submarine

**Hardware type**
ArduSub SITL commit https://github.com/ArduPilot/ardupilot/tree/9f1c4df5e7

srmainwaring, 2023-02-07:
Closing as needs to be resolved upstream.

closed 2023-02-07 by srmainwaring

Rezenders, 2023-02-07:
Wow, thanks for the detailed analyzes and information @srmainwaring!
I really appreciate it.