ArduPilot/ardupilot_gazebo #46: undefined symbol

eppravitra, 2023-02-12:
## Bug report

**Issue details**

I couldn't solve this link error. The plugin at commit 5b4d5a7 compiled just fine
```
$ gz sim -r iris_runway.sdf
Error while loading the library [/home/ep/ardupilot_gazebo/build/libArduPilotPlugin.so]: /home/ep/ardupilot_gazebo/build/libArduPilotPlugin.so: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev, version Qt_5
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [ArduPilotPlugin]
- Requested library name: [libArduPilotPlugin]
- Resolved library path: [/home/ep/ardupilot_gazebo/build/libArduPilotPlugin.so]
```
**Version**
ardupilot_gazebo commit 5b4d5a7
Gazebo Garden
Ubuntu 22.04
gcc 12.1.0

Any help would be greatly appreciated.

srmainwaring, 2023-02-12:
@eppravitra what was the last commit you successfully built? (you've provided the same commit twice above).  

The most recent changes have not altered the C++ code - an unused file was removed, but otherwise the changes affect the SDF model files not the plugin code.

I built a clean Ubuntu 22.04 VM hosted on macOS VMware 13.0.1. Gazebo Garden is a binary install. All packages updated and upgraded. The ardupilot plugin builds clean. The iris runway world launches and runs as expected.

<details>

```bash
$ uname -a
Linux ubuntu-jammy 5.15.0-60-generic #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

```bash
$ gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

```bash
$ gz sim --version
Gazebo Sim, version 7.3.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.
```

```bash
rhys@ubuntu-jammy:~/Code/ardupilot/sim/ardupilot_gazebo/build$ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebugInfo -DCMAKE_CXX_STANDARD=17
-- Looking for gz-sim7 -- found version 7.3.0
-- Searching for dependencies of gz-sim7
-- Looking for sdformat13 -- found version 13.3.0
-- Searching for dependencies of sdformat13
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for gz-math7 -- found version 7.1.0
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.0.0
-- Searching for dependencies of gz-utils2-cli
-- Looking for gz-plugin2 -- found version 2.0.1
-- Searching for dependencies of gz-plugin2
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for <gz-plugin2> component [loader]
-- Looking for gz-plugin2-loader -- found version 2.0.1
-- Searching for dependencies of gz-plugin2-loader
-- Searching for <gz-plugin2> component [register]
-- Looking for gz-plugin2-register -- found version 2.0.1
-- Searching for dependencies of gz-plugin2-register
-- Looking for gz-transport12 -- found version 12.1.0
-- Searching for dependencies of gz-transport12
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Looking for gz-utils2 -- found version 2.0.0
-- Looking for gz-msgs9 -- found version 9.3.0
-- Searching for dependencies of gz-msgs9
-- Looking for gz-math7 -- found version 7.1.0
-- Searching for <gz-transport12> component [log]
-- Looking for gz-transport12-log -- found version 12.1.0
-- Searching for dependencies of gz-transport12-log
-- Searching for <gz-transport12> component [parameters]
-- Looking for gz-transport12-parameters -- found version 12.1.0
[... 165 lines cut ...]
-- Looking for gz-sim7 - found

-- RapidJSON found. Headers: /usr/include
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rhys/Code/ardupilot/sim/ardupilot_gazebo/build
rhys@ubuntu-jammy:~/Code/ardupilot/sim/ardupilot_gazebo/build$ make
[ 16%] Building CXX object CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o
[ 33%] Building CXX object CMakeFiles/ArduPilotPlugin.dir/src/Socket.cpp.o
[ 50%] Building CXX object CMakeFiles/ArduPilotPlugin.dir/src/Util.cc.o
[ 66%] Linking CXX shared library libArduPilotPlugin.so
[ 66%] Built target ArduPilotPlugin
[ 83%] Building CXX object CMakeFiles/ParachutePlugin.dir/src/ParachutePlugin.cc.o
[100%] Linking CXX shared library libParachutePlugin.so
[100%] Built target ParachutePlugin
rhys@ubuntu-jammy:~/Code/ardupilot/sim/ardupilot_gazebo/build$ gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

</details>

The link error you are getting seems to be a problem with your std library / gcc installation. Your version of gcc is different to the one I have from a standard install - that may be causing runtime library conflicts. All I can suggest is to check that all dependent packages are updated and upgraded. AFAICT there does not appear to be a problem with the plugin on Ubuntu Jammy.

eppravitra, 2023-02-13:
Hi @srmainwaring 
The last commit that I got it to work was 1baf84290a3af63a0ce92bc4d8af2696d5940045. It was right before the upgrade from Fotress to Garden.

Thanks for the suggestion. I'll try to hunt down what's going on with my std library and gcc.

eppravitra, 2023-02-13:
Hi @srmainwaring 

I got it work! Thank you! 

Unfortunately, I have no idea what I did that really got it to work. I think it's because I uninstall all gcc and g++ versions and then reinstall gcc/g++12.

I can confirm that the plugin works for gcc12.