ArduPilot/ardupilot_gazebo #61: CMake error

IAMHAADICOOL, 2023-06-21:
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findgz-cmake3.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "gz-cmake3", but CMake did not find one.

  Could not find a package configuration file provided by "gz-cmake3" with
  any of the following names:

    gz-cmake3Config.cmake
    gz-cmake3-config.cmake

  Add the installation prefix of "gz-cmake3" to CMAKE_PREFIX_PATH or set
  "gz-cmake3_DIR" to a directory containing one of the above files.  If
  "gz-cmake3" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

IAMHAADICOOL, 2023-06-21:
How do I fix this?

srmainwaring, 2023-06-21:
Hello @IAMHAADICOOL, it could be a number of things depending upon your OS and whether you install Gazebo from source or binaries.

- First check you have installed Gazebo Garden (`gz sim --version` should be 7.x.x) 

- If you have a binary install, you may be missing the `*-dev` packages that include headers etc. `sudo apt install libgz-sim7-dev`.

- If you have a source install ensure that you have sourced the `./install/setup.bash` in the `gz-sim` workspace before building the plugin.

srmainwaring, 2023-07-27:
Closing as no further activity.

closed 2023-07-27 by srmainwaring