ArduPilot/ardupilot_gazebo #71: [GUI] [Dbg] [Application.cc:242] Terminating application.

metaphaz, 2023-12-22:
## Bug report

I just installed gazebo from the tutorial from ardupilot but i cant lunch gazebo with "gz sim -v4 -r iris_runway.sdf"  or "gz sim -v4 -r zephyr_runway.sdf"

**Version**
Ubuntu 22.04.3
Gazebo Sim, version 8.0.0
ArduCopter 4.4.4
ArduPLane 4.1

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
İris and Zephyr

**Logs**
"[Msg] Gazebo Sim GUI    v8.0.0
[Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:263] Waiting for subscribers to [/gazebo/starting_world]...
[Msg] Received world [iris_runway.sdf] from the GUI.
[Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].
[Msg] Gazebo Sim Server v8.0.0
[Msg] Loading SDF world file[/home/metaphaz/gz_ws/src/ardupilot_gazebo/worlds/iris_runway.sdf].
[Dbg] [Application.cc:101] Initializing application.
[Dbg] [Application.cc:175] Qt using OpenGL graphics interface
[GUI] [Dbg] [Application.cc:662] Create main window
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[Dbg] [Sensors.cc:511] Configuring Sensors system
[Dbg] [Sensors.cc:430] SensorsPrivate::Run
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Sensors] for entity [1]
[Dbg] [Sensors.cc:406] SensorsPrivate::RenderThread started
[Dbg] [Sensors.cc:246] Waiting for init
[Msg] Create service on [/world/iris_runway/create]
[Msg] Remove service on [/world/iris_runway/remove]
[Msg] Pose service on [/world/iris_runway/set_pose]
[Msg] Pose service on [/world/iris_runway/set_pose_vector]
[Msg] Light configuration service on [/world/iris_runway/light_config]
[Msg] Physics service on [/world/iris_runway/set_physics]
[Msg] SphericalCoordinates service on [/world/iris_runway/set_spherical_coordinates]
[Msg] Enable collision service on [/world/iris_runway/enable_collision]
[Msg] Disable collision service on [/world/iris_runway/disable_collision]
[Msg] Material service on [/world/iris_runway/visual_config]
[Msg] Material service on [/world/iris_runway/wheel_slip]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Imu] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::NavSat] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::JointStatePublisher] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_0_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_1_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_2_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_3_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[libprotobuf ERROR google/protobuf/descriptor_database.cc:175] Symbol name "gz.sim.private_msgs.PerformerAffinity" conflicts with the existing symbol "gz.sim.private_msgs.PerformerAffinity".
[libprotobuf FATAL google/protobuf/descriptor.cc:1382] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7fad7c914c00, in gz::sim::v8::SdfEntityCreator::CreateEntities(sdf::v14::Model const*)
#30   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7fad7c8ecf55, in 
#29   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7fad7c94d268, in gz::sim::v8::SimulationRunner::LoadPlugins(unsigned long, std::vector<sdf::v14::Plugin, std::allocator<sdf::v14::Plugin> > const&)
#28   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7fad7c959d33, in gz::sim::v8::SystemManager::LoadPlugin(unsigned long, sdf::v14::Plugin const&)
#27   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7fad7c9594e9, in gz::sim::v8::SystemLoader::LoadPlugin(sdf::v14::Plugin const&)
#26   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7fad7c9584a8, in 
#25   Object "/lib/x86_64-linux-gnu/libgz-plugin2-loader.so.2", at 0x7fad7cf5e56f, in gz::plugin::Loader::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
#24   Object "/lib/x86_64-linux-gnu/libgz-plugin2-loader.so.2", at 0x7fad7cf5d4dd, in gz::plugin::Loader::Implementation::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
#23   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81a906c7, in dlopen
#22   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81a9012d, in 
#21   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81b74972, in _dl_catch_error
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81b748a7, in _dl_catch_exception
#19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81a9063b, in 
#18   Object "/lib64/ld-linux-x86-64.so.2", at 0x7fad822a634d, in 
#17   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81b748a7, in _dl_catch_exception
#16   Object "/lib64/ld-linux-x86-64.so.2", at 0x7fad822a5ff5, in 
#15   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81b74904, in _dl_catch_exception
#14   Object "/lib64/ld-linux-x86-64.so.2", at 0x7fad8229e567, in 
#13   Object "/lib64/ld-linux-x86-64.so.2", at 0x7fad8229e47d, in 
#12   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7fad490b3d08, in 
#11   Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7fad7bd81951, in google::protobuf::internal::AddDescriptors(google::protobuf::internal::DescriptorTable const*)
#10   Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7fad7bd81966, in google::protobuf::internal::AddDescriptors(google::protobuf::internal::DescriptorTable const*)
#9    Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7fad7bc92c35, in 
#8    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7fad81c4e2dc, in _Unwind_Resume
#7    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7fad81c4d883, in 
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fad7d8ad958, in __gxx_personality_v0
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fad7d8ad1e8, in 
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fad7d8ae20b, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fad7d8a2b9d, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81a287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81a42475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fad81a969fc, in pthread_kill
Aborted (Signal sent by tkill() 43776 1000)
[GUI] [Dbg] [PathManager.cc:68] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [SignalHandler.cc:142] Received signal[2].
[GUI] [Dbg] [SignalHandler.cc:142] Received signal[2].
[GUI] [Dbg] [Application.cc:242] Terminating application."

srmainwaring, 2023-12-22:
> **Version**
> Ubuntu 22.04.3
> Gazebo Sim, version 8.0.0
> ArduCopter 4.4.4
> ArduPLane 4.1

You are using Gazebo Harmonic - we have only just added support for this so I'm hoping there are not teething issues on Ubuntu.

Did you set the environment variable `GZ_VERSION=harmonic` before building the plugin, and do you have Gazebo Garden installed anywhere as well? It looks like a conflict between the protobuf dependencies of `gz-msgs9` vs `gz-msgs10`.

metaphaz, 2023-12-22:
- No i did not set environment variable as `GZ_VERSION=harmonic` 
- Yes i have Garden and Harmonic together
Should i repeat installation by setting environment variable as `GZ_VERSION=harmonic` ?

Thank you for your fast answer.

srmainwaring, 2023-12-23:
> Should i repeat installation by setting environment variable as GZ_VERSION=harmonic

Yes, and make sure that you don't have any of the Gazebo Garden libraries in the `GZ_SIM_SYSTEM_PLUGIN` path as it's an ABI conflict that you are seeing.

srmainwaring, 2023-12-27:
@metaphaz - did you manage to resolve your issue with a clean environment?

srmainwaring, 2024-01-08:
Closing due to inactivity.

closed 2024-01-08 by srmainwaring

Mike-Solar, 2024-01-27:
same problem.