ArduPilot/ardupilot_gazebo #72: Multi Vehicle SITL support

efekrtt, 2023-12-26:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

efekrtt, 2023-12-26:
I want to control multi vehicle from Ardupilot Gazebo. Is there any resource I can use ?

srmainwaring, 2023-12-27:
Hi @efekrtt, this is a user support question rather than a software issue. A better forum for this is https://discuss.ardupilot.org/ where you may find threads discussing multi-vehicle support in SITL with Gazebo, such as this one: https://discuss.ardupilot.org/t/ship-landing-ignition-simulation/80663/4

closed 2023-12-27 by srmainwaring