ArduPilot/ardupilot_gazebo #77: Runtime Crash with protobuf

Ryanf55, 2024-01-20:
## Bug report

**Issue details**

Trying to build+launch gazebo harmonic with the `gimbal.sdf` world has a crash on Ubuntu 22.

**Version**
commit 1caa507896faf758eb52e12d47fa0c192777f114 (origin/ros2, ros2)

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**
```
$ gz sim -r gimbal.sdf
[libprotobuf ERROR google/protobuf/descriptor_database.cc:175] Symbol name "gz.sim.private_msgs.PerformerAffinity" conflicts with the existing symbol "gz.sim.private_msgs.PerformerAffinity".
[libprotobuf FATAL google/protobuf/descriptor.cc:1382] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f6d4ad14c60, in gz::sim::v8::SdfEntityCreator::CreateEntities(sdf::v14::Model const*)
#30   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f6d4acecf55, in 
#29   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f6d4ad4d268, in gz::sim::v8::SimulationRunner::LoadPlugins(unsigned long, std::vector<sdf::v14::Plugin, std::allocator<sdf::v14::Plugin> > const&)
#28   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f6d4ad59d33, in gz::sim::v8::SystemManager::LoadPlugin(unsigned long, sdf::v14::Plugin const&)
#27   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f6d4ad594e9, in gz::sim::v8::SystemLoader::LoadPlugin(sdf::v14::Plugin const&)
#26   Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f6d4ad584a8, in 
#25   Object "/lib/x86_64-linux-gnu/libgz-plugin2-loader.so.2", at 0x7f6d4b2da56f, in gz::plugin::Loader::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
#24   Object "/lib/x86_64-linux-gnu/libgz-plugin2-loader.so.2", at 0x7f6d4b2d94dd, in gz::plugin::Loader::Implementation::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
#23   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f6906c7, in dlopen
#22   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f69012d, in 
#21   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f774b62, in _dl_catch_error
#20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f774a97, in _dl_catch_exception
#19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f69063b, in 
#18   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f6d4fe4234d, in 
#17   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f774a97, in _dl_catch_exception
#16   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f6d4fe41ff5, in 
#15   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f774af4, in _dl_catch_exception
#14   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f6d4fe3a567, in 
#13   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f6d4fe3a47d, in 
#12   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f6d072b3d08, in 
#11   Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7f6d4a181951, in google::protobuf::internal::AddDescriptors(google::protobuf::internal::DescriptorTable const*)
#10   Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7f6d4a181966, in google::protobuf::internal::AddDescriptors(google::protobuf::internal::DescriptorTable const*)
#9    Object "/lib/x86_64-linux-gnu/libprotobuf.so.23", at 0x7f6d4a092c35, in 
#8    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7f6d4bffa2dc, in _Unwind_Resume
#7    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7f6d4bff9883, in 
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f6d4bcad958, in __gxx_personality_v0
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f6d4bcad1e8, in 
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f6d4bcae20b, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f6d4bca2b9d, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f6287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f642475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6d4f6969fc, in pthread_kill
Aborted (Signal sent by tkill() 320400 1000)
```

**Environment**

Ubuntu 22.04. 
```
ryan@B650-970:~/Dev/ros2_ws/src/ardupilot_gazebo$ echo $GZ_VERSION 
harmonic
ryan@B650-970:~/Dev/ros2_ws/src/ardupilot_gazebo$ echo $GZ_SIM_RESOURCE_PATH 
/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/models:/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/worlds:/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/models:/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/worlds:
ryan@B650-970:~/Dev/ros2_ws/src/ardupilot_gazebo$ echo $GZ_SIM_SYSTEM_PLUGIN_PATH 
/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/build:/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/build:
```

davidbitton, 2024-01-21:
@Ryanf55 i'm seeing an error when attempting to start sim w/ `iris_runway.sdf`.
```
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1705794792.190421  795402 descriptor_database.cc:633] File already exists in database: gz/msgs/time.proto
F0000 00:00:1705794792.190474  795402 descriptor.cc:2122] Check failed: GeneratedDatabase()->Add(encoded_file_descriptor, size)
```

I omitted the stack trace. I'm hoping it's the "same" issue.

Ryanf55, 2024-01-21:
Thanks for the extra info. I'm working with the Gazebo folks to get a debug build of gazebo. It doesn't yet compile on my dev machine (Ubuntu 22).
https://github.com/gazebosim/gz-rendering/issues/962

Once it does, we should be able to chase it down quick in the debugger.

srmainwaring, 2024-01-21:
@Ryanf55 is this an ArduPilot plugin issue or can you reproduce it running the one of the Gazebo Sim examples?

Conflicting  protobuf libraries have come up before;
-  https://github.com/gazebosim/gz-sim/issues/1990

If you have more than one version of Gazebo installed concurrently that can cause problems. It's also a known issue with Gazebo that multiple versions of Python libraries are linked against (different libraries require different Python versions).

With this type of problem debugging may be less useful, as what generally happens is you break in one of the `dlopen()` (or linux equivalent) statements, probably in the plugin loader. I'd start with running

```bash
LD_TRACE_LOADED_OBJECTS=1 gz sim -v4 -r gimbal.sdf
```

or one of the other library load debug commands recommended here: https://stackoverflow.com/questions/22784635/equivalent-of-dyld-print-libraries-on-linux
 
and inspect what gets loaded from where. The equivalent on macOS is 

```bash
DYLD_PRINT_LIBRARIES = 1 gz sim -v4 -r gimbal.sdf
```

davidbitton, 2024-01-21:
@srmainwaring, at least for me, I can compile the ardupilot plugin. This failure is at runtime. I _think_ it's the Adrupilot plugin and the exception is misleading. I need to debug the code.

Ryanf55, 2024-01-21:
> ```shell
> LD_TRACE_LOADED_OBJECTS=1 gz sim -v4 -r gimbal.sdf
> ```

After building the `ros2` branch in a clean terminal with `GZ_VERSION=harmonic`:

```bash
ryan@B650-970:~/Dev/ros2_ws/src/ardupilot_gazebo$ gz sim -r gimbal.sdf
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
Stack trace (most recent call last):
#29   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#28   Object "gz sim server", at 0x5640bf1c61c4, in _start
#27   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1711e29e3f, in __libc_start_main
#26   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1711e29d8f, in 
#25   Object "gz sim server", at 0x5640bf1c617e, in 
#24   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f17122a8e19, in ruby_run_node
#23   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f17122a5317, in 
#22   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f171243a30c, in rb_vm_exec
#21   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f1712434c96, in 
#20   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f1712431fc5, in 
#19   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f171242fc34, in 
#18   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f171237ba1e, in 
#17   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f17122a69ac, in rb_protect
#16   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f171243ec61, in rb_yield
#15   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f171243a30c, in rb_vm_exec
#14   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f1712434c96, in 
#13   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f1712431fc5, in 
#12   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f171242fc34, in 
#11   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7f17125a744b, in 
#10   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f17123fd088, in rb_nogvl
#9    Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7f17125a6d6b, in 
#8    Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f1712598492, in 
#7    Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f171259be2d, in 
#6    Object "/usr/lib/x86_64-linux-gnu/libgz-sim8-gz.so.8.0.0", at 0x7f17120757b2, in runServer
#5    Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f170d137cf5, in 
#4    Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f170d1437c2, in gz::sim::v8::SimulationRunner::Run(unsigned long)
#3    Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f170d14fd10, in gz::sim::v8::SimulationRunner::Step(gz::sim::v8::UpdateInfo const&)
#2    Object "/lib/x86_64-linux-gnu/libgz-sim8.so.8", at 0x7f170d142bb1, in gz::sim::v8::SimulationRunner::UpdateSystems()
#1    Object "/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/build/libCameraZoomPlugin.so", at 0x7f16c9882aa9, in gz::sim::v8::systems::CameraZoomPlugin::PreUpdate(gz::sim::v8::UpdateInfo const&, gz::sim::v8::EntityComponentManager&)
#0    Object "/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/build/libCameraZoomPlugin.so", at 0x7f16c988125c, in gz::sim::v8::systems::CameraZoomPlugin::Impl::InitialiseCamera()
Segmentation fault (Address not mapped to object [(nil)])
ryan@B650-970:~/Dev/ros2_ws/src/ardupilot_gazebo$ gz sim -v 4 -r gimbal.sdf
[Msg] Gazebo Sim GUI    v8.0.0
[Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:263] Waiting for subscribers to [/gazebo/starting_world]...
[Msg] Received world [gimbal.sdf] from the GUI.
[Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].
[Msg] Gazebo Sim Server v8.0.0
[Msg] Loading SDF world file[/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/worlds/gimbal.sdf].
[Dbg] [Application.cc:101] Initializing application.
[Dbg] [Application.cc:175] Qt using OpenGL graphics interface
[GUI] [Dbg] [Application.cc:662] Create main window
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[Dbg] [Sensors.cc:511] Configuring Sensors system
[Dbg] [Sensors.cc:430] SensorsPrivate::Run
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Sensors] for entity [1]
[Dbg] [Sensors.cc:406] SensorsPrivate::RenderThread started
[Dbg] [Sensors.cc:246] Waiting for init
[Msg] Create service on [/world/gimbal/create]
[Msg] Remove service on [/world/gimbal/remove]
[Msg] Pose service on [/world/gimbal/set_pose]
[Msg] Pose service on [/world/gimbal/set_pose_vector]
[Msg] Light configuration service on [/world/gimbal/light_config]
[Msg] Physics service on [/world/gimbal/set_physics]
[... 654 lines cut ...]
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[... cut ...]
Both `d` and `Tr` parameters defined for "Wheels3". Use the value of `d` for dissolve (line 22 in .mtl.)
Both `d` and `Tr` parameters defined for "Hybrid_Interior". Use the value of `d` for dissolve (line 36 in .mtl.)
Both `d` and `Tr` parameters defined for "Windows". Use the value of `d` for dissolve (line 50 in .mtl.)

[GUI] [Dbg] [Gui.cc:523] Shutting down gz-sim-gui
[GUI] [Dbg] [Application.cc:242] Terminating application.
[GUI] [Msg] Loading plugin [gz-rendering-ogre2]
[GUI] [Dbg] [MinimalScene.cc:838] Destroy scene [scene]
```
```

srmainwaring, 2024-01-21:
@Ryanf55 if you run the server and gui separately which process seg faults? (forced to do this on macOS, but it's a useful approach for tracing issues such as this).

Ryanf55, 2024-01-21:
> @Ryanf55 if you run the server and gui separately which process seg faults? (forced to do this on macOS, but it's a useful approach for tracing issues such as this).
> 
> You're getting a very different type of error this time - not protobuf, but looks like something with rendering?
> 
> ```shell
> #1    Object "/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/build/libCameraZoomPlugin.so", at 0x7f16c9882aa9, in gz::sim::v8::systems::CameraZoomPlugin::PreUpdate(gz::sim::v8::UpdateInfo const&, gz::sim::v8::EntityComponentManager&)
> #0    Object "/home/ryan/Dev/ros2_ws/src/ardupilot_gazebo/build/libCameraZoomPlugin.so", at 0x7f16c988125c, in gz::sim::v8::systems::CameraZoomPlugin::Impl::InitialiseCamera()
> Segmentation fault (Address not mapped to object [(nil)])
> ```

It crashes in headless mode too, so not rendering.
```bash
 gz sim -s -v4 -r gimbal.sdf

```

Ryanf55, 2024-01-21:
Through some handy printf debugging, it's crashing on the call to: 
```
this->scene = rendering::sceneFromFirstRenderEngine();
```

srmainwaring, 2024-01-21:
Ok - that starts to narrow things down, so rendering.

```bash
Segmentation fault (Address not mapped to object [(nil)])
```

Is something I see in a VM when trying to run with hardware acceleration and there is an issue with OpenGL. It may be unconnected, but in my case it points to a problem with the OpenGL drivers (which in the case of OpenGL on VMware still do not support ogre2 rendering with acceleration even though the mesa version claims support for OpenGL 4.3).

I can see if I can replicate in an Ubuntu VM with software rendering . In the meanwhile you could enable the LIBGL_DEBUG debug flag to see if it raises anything: https://docs.mesa3d.org/envvars.html

Ryanf55, 2024-01-22:
> Ok - that starts to narrow things down, so rendering.
> 
> ```shell
> Segmentation fault (Address not mapped to object [(nil)])
> ```
> 
> Is something I see in a VM when trying to run with hardware acceleration and there is an issue with OpenGL. It may be unconnected, but in my case it points to a problem with the OpenGL drivers (which in the case of OpenGL on VMware still do not support ogre2 rendering with acceleration even though the mesa version claims support for OpenGL 4.3).
> 
> I can see if I can replicate in an Ubuntu VM with software rendering . In the meanwhile you could enable the LIBGL_DEBUG debug flag to see if it raises anything: https://docs.mesa3d.org/envvars.html
> 
> I'm assuming the scene pointer is valid?

```
[Dbg] [JointPositionController.cc:361] Identified joint [gimbal::yaw_joint] as Entity [32]
[Dbg] [JointPositionController.cc:361] Identified joint [gimbal::roll_joint] as Entity [33]
[Dbg] [JointPositionController.cc:361] Identified joint [gimbal::pitch_joint] as Entity [34]
[Err] [CameraZoomPlugin.cc:168] AAB
[Err] [CameraZoomPlugin.cc:171] AAC
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [yaw_collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [roll_collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [pitch_collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Dbg] [SDFFeatures.cc:328] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[Dbg] [SDFFeatures.cc:843] The geometry element of collision [person] couldn't be created
[Dbg] [CameraZoomPlugin.cc:403] [Dbg] [Sensors.cc:682] Initialization needed
[Dbg] [CameraZoomPlugin.cc:403] [Dbg] [Sensors.cc:682] Camera name: [mount::gimbal::pitch_link::camera].
[Dbg] [Sensors.cc:258] Initializing render context
[Msg] Loading plugin [gz-rendering-ogre2]
[Msg] Serving scene information on [/world/gimbal/scene/info]
[Msg] Serving graph information on [/world/gimbal/scene/graph]
[Msg] Serving full state on [/world/gimbal/state]
[Msg] Serving full state (async) on [/world/gimbal/state_async]
[Msg] Publishing scene information on [/world/gimbal/scene/info]
[Msg] Publishing entity deletions on [/world/gimbal/scene/deletion]
[Msg] Publishing state changes on [/world/gimbal/state]
[Msg] Publishing pose messages on [/world/gimbal/pose/info]
[Msg] Publishing dynamic pose messages on [/world/gimbal/dynamic_pose/info]
[Dbg] [EntityComponentManager.cc:1655] Updated state thread iterators: 32 threads processing around 5 entities each.
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Model.hh:69] [Wrn] [Model.hh:69] [Wrn] [Model.hh:69] Skipping serialization / deserialization for models 
Skipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Model.hh:69] [Wrn] [Model.hh:69] [Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.

Trying to serialize component with data type [N3sdf3v145WorldE], which doesn't have `operator<<`. Component will not be serialized.
[Err] [CameraZoomPlugin.cc:168] AAB
[GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v145WorldE], which doesn't have `operator>>`. Component will not be deserialized.
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[... cut ...]
I added an assertion just after this call that the scen is not null, and it returns false. Thus, the logic of calling `!this->scene->IsInitialized()` is dereferencing a nullptr. I should have known, printf debugging would be dropped just before a segfault because IO doesn't have time to print to console before the program is halted.

I was able to make a fix to this and pushed a patch to my branch `bugfix/camera-initialize-nullptr-crash.` While it protects against the crash, the camera doesn't seem to ever retry. I can't find where `InitialiseCamera` gets called. The topic exists, but there is no data. I have to manually add the image display to get that topic box to show up. 

[image]

srmainwaring, 2024-01-22:
Thanks for the fix, missed the check for the scene ptr before checking it's valid.

You can add the `ImageDisplay` widget to your gui.config, or you can save your config when it's loaded using the left menu. The sensors in Gazebo are lazy, they will not publish unless there is a  subscriber. Not sure whether this also includes initialisation?

`InitialiseCamera` is called in the plugin Pre-Update. It will be called every loop until the member variable `camera`, which is set in the function, is valid.

Ryanf55, 2024-01-22:
> Thanks for the fix, missed the check for the scene ptr before checking it's valid.
>
> You can add the `ImageDisplay` widget to your gui.config, or you can save your config when it's loaded using the [...]

The reason it wasn't showing up was because I needed to hit play :facepalm: 

PR submitted!

PR #79 (2024-01-25): Add protection for nulltpr crash in CameraZoomPlugin
Fixes #77 

Make sure to hit Play in gazebo for the image topic to show up! 

[image]
closed 2024-01-25 by srmainwaring