ArduPilot/ardupilot_gazebo #78: Zephyr: update model to conform to ROS REP 105

srmainwaring, 2024-01-21:
## Feature request

**Is your feature request related to a problem? Please describe.**

- The `zephyr` model does not conform to [ROS REP 105](https://www.ros.org/reps/rep-0105.html).
- This is a problem as users may take the `zephyr` as a template for adding new models, and propagate the unconventional orientation (for and example see: https://github.com/ArduPilot/SITL_Models/pull/116).

**Describe the solution you'd like**

- Update the `zephyr` model to conform to [ROS REP 105](https://www.ros.org/reps/rep-0105.html).

**Describe alternatives you've considered**

- Leave the model as is, which is consistent with the original: https://app.gazebosim.org/OpenRobotics/fuel/models/Zephyr%20Delta%20Wing, which is unfortunately also wrong.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

Hilalbasibuyuk, 2024-04-17:
I want to find PLANE model but there is no anywhere, also the plane model should have camera plugin

srmainwaring, 2024-04-17:
@Hilalbasibuyuk, this issue is specific to the Zephyr model, which is available in the repo but has a non-standard convention for the orientation of it's x and y axis.

The zephyr is simulated using ArduPilot Plane, so if you require a plane then this will be suitable. Otherwise you can look in ArduPilot/SITL_Models for other examples. There is a PR for a Mini Talon v-tail. If you have a specific vehicle in mind please open a new issue in ArduPilot/SITL_Models or open a topic on ArduPilot Discourse.