ArduPilot/ardupilot_gazebo #84: installation error

jinesh793, 2024-02-04:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

i am trying to install the plugins for the ardupilot  on mac m1 gazebo i am facing the error for the incomplete configuration i need an insigth on it 
" ardupilot_gazebo % cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo                                                                    
CMake Error at CMakeLists.txt:37 (find_package):
  By not providing "Findgz-cmake3.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "gz-cmake3", but CMake did not find one.

  Could not find a package configuration file provided by "gz-cmake3" with
  any of the following names:

    gz-cmake3Config.cmake
    gz-cmake3-config.cmake

  Add the installation prefix of "gz-cmake3" to CMAKE_PREFIX_PATH or set
  "gz-cmake3_DIR" to a directory containing one of the above files.  If
  "gz-cmake3" provides a separate development package or SDK, be sure it has
  been installed."

_Please describe the problem_

**Version**
_What version was the issue encountered with_ gazebo11
**Platform**
[  x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

\

srmainwaring, 2024-02-04:
Hi @jinesh793, this look like a configuration issue in your build setup. Could you please remove the top four lines of the issue as well.

Can you provide some more information about your system:

- Have you installed `gz-sim` from source or from the binaries?
- Current `gz-sim` version: (what does `gz sim --version` output?).

I develop on a M1 MacBook, but build Gazebo from source and ensure that the workspace used to build Gazebo is sourced before building the plugin. This configures the environment variables required for plugin build. With a binary install you may need to manually update `CMAKE_PREFIX_PATH` to include the packages required in the plugin `CMakeLists.txt` as they will not be set automatically.

srmainwaring, 2024-02-16:
Closed due to lack of follow-up / activity.

closed 2024-02-16 by srmainwaring