ArduPilot/ardupilot_gazebo #89: Add multi-agent heterogenous demo with zephyr and iris in one environment

Ryanf55, 2024-04-18:
## Feature request

**Is your feature request related to a problem? Please describe.**

I want to simulate a mixed fleet of vehicle types in Gazebo, but it's not clear how to do that.

**Describe the solution you'd like**

https://discuss.ardupilot.org/t/multi-vehicle-zephyr-and-iris-ardupilot-gazebo-simulation-going-to-different-points-in-gazebo-simulation-for-the-same-lat-lon-alt/90686

Add the iris and the zephyr models to the same environment, as requested above. Then, demonstrate how to run SITL where you can have two mavproxy sessions, one for each vehicle, and have both vehicles flying at the same time (perhaps hover the iris, and then have the plane do a loiter circle around).

**Describe alternatives you've considered**

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Additional context**

srmainwaring, 2024-04-20:
@Ryanf55 - this is already supported and used in the ship landing demos where a quadcopter and truck or quadcopter and ship are both controller in Gazebo using ArduPilot. There are a couple of threads in discourse with discussions:

- https://discuss.ardupilot.org/t/ship-landing-ignition-simulation/80663/19
- https://discuss.ardupilot.org/t/combine-two-working-models-for-gz-sim/111313/10

Ryanf55, 2024-04-20:
Thanks for the references. I'm happy to do the work for it with the content you attached.

srmainwaring, 2024-05-24:
> What do you think about adding a node that makes the copter with gimbal chase the plane and film it with the gimbal?

This would be better handled in https://github.com/ArduPilot/ardupilot_gz where we can use Python launch files to manage multiple vehicles (and override the port settings in the additional vehicles).

srmainwaring, 2026-04-05:
Resolved by https://github.com/ArduPilot/ardupilot_gz/pull/67