ArduPilot/ardupilot_gazebo #95: Unable to install this plugin on M1 Mac using GZ Harmonic

nacorti, 2024-06-06:
## Bug report

**Issue details**
Attempting to install this plugin on an M1 Macbook Pro, running OSX 14.4.1. However, I persistently run into this error message

```
CMake Error at /opt/homebrew/lib/cmake/protobuf/protobuf-targets.cmake:71 (set_target_properties):
  The link interface of target "protobuf::libprotobuf" contains:

    absl::absl_check

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/protobuf/protobuf-config.cmake:16 (include)
  /opt/homebrew/share/cmake/gz-cmake3/cmake3/FindGzProtobuf.cmake:29 (find_package)
  /opt/homebrew/lib/cmake/gz-transport13/gz-transport13-config.cmake:92 (find_package)
  /opt/homebrew/lib/cmake/gz-sim8/gz-sim8-config.cmake:94 (find_package)
  /opt/homebrew/share/cmake/gz-cmake3/cmake3/GzFindPackage.cmake:243 (find_package)
  CMakeLists.txt:31 (gz_find_package)
```

_Please describe the problem_
I've installed GZ harmonic using `brew install gz-harmonic`. I'm able to run simple test examples using just `gazebo`. I've exported the `GZ_VERSION` env variable. However running `cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo` as instructed in the install notes causes the above error. I can see that `brew` is pulling down `protobuf` v27 and `abseil` successfully, but something in this installation isn't picking up `abseil` properly. Wondering if you guys have certain versions of these dependency libraries that you pin that work for y'all. I'm in the process of going down the rabbit hole to try and get an earlier version of `protobuf` without `abseil` to work but figured I'd ask around here. I'm hoping at least one of y'all has this working on an M1 Mac.

Thanks in advance!

**Version**
_What version was the issue encountered with_
GZ Harmonic, attempting to build this package from latest master
**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**
_Please provide a link to any relevant logs that show the issue_

srmainwaring, 2024-06-06:
@nacorti - the ardupilot gazebo plugin does not use protobuf or abseil directly as it does not need to generate custom message types. I wonder whether the versions used to build gz-harmonic are of date compared to those available in brew latest in you set up (will need to look at the brew formula for osrf and compare to `brew info protobuf` for this)?

I build `gz-harmonic` from source on macOS and don't see this issue, because the the gz libraries and additional plugins are always in sync. If you could try a source build for comparison that would be helpful.

nacorti, 2024-06-06:
@srmainwaring Understood, I'll try a source build tonight and report back 🫡

srmainwaring, 2024-06-24:
Closing as no follow up activity.

closed 2024-06-24 by srmainwaring