ArduPilot/ardupilot_gazebo #105: Gazebo simulation doesn't start with `gimbal.sdf` world

srizvan, 2024-07-07:
## Bug report

**Issue details**

Gazebo simulation fails to run with `gimbal.sdf` world. It throws the next error:
```[Msg] Gazebo Sim Server v8.5.0
[Msg] Loading SDF world file[/Users/sergiir/git/gz_ws/src/ardupilot_gazebo/worlds/gimbal.sdf].
[Err] [Server.cc:203] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [file://media/materials/scripts/gazebo.material] could not be resolved.
[Err] [Server.cc:203] Error Code 9: Msg: Failed to load a world.
[Dbg] [gz.cc:434] Shutting down gz-sim-server```

Other worlds don't have any issues.

**Version**
Gazebo Harmonic
MacOS Sonoma 14.5

**Platform**
[] Copter

IuliuNovac, 2024-07-08:
I am getting the same error.

IuliuNovac, 2024-07-08:
This seems to be the root of the issue, i have removed it and it somehow works.
`  
  <include>
    <uri>https://fuel.gazebosim.org/1.0/OpenRobotics/models/Prius Hybrid</uri>
    <name>prius_hybrid0</name>
    <pose degrees="true">0 7 0 0 0 40</pose>
  </include>`

IuliuNovac, 2024-07-08:
The example doesn't work anyway, what a pain. 
`gst-launch-1.0 -v udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink sync=false`

srizvan, 2024-07-08:
It seems that gstreamer plugin requests Protobuf 27.0, but I have 27.1 installed:

```Error while loading the library [/Users/sergiir/git/gz_ws/src/ardupilot_gazebo/build/libGstCameraPlugin.dylib]: dlopen(/Users/sergiir/git/gz_ws/src/ardupilot_gazebo/build/libGstCameraPlugin.dylib, 0x0085): Library not loaded: /usr/local/opt/protobuf/lib/libprotobuf.27.0.0.dylib
  Referenced from: <FE1FFA60-735C-3F5A-BB55-F1BC5247B1CD> /usr/local/Cellar/opencv/4.9.0_10/lib/libopencv_dnn.4.9.0.dylib
  Reason: tried: '/usr/local/opt/protobuf/lib/libprotobuf.27.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/protobuf/lib/libprotobuf.27.0.0.dylib' (no such file), '/usr/local/opt/protobuf/lib/libprotobuf.27.0.0.dylib' (no such file), '/usr/local/Cellar/protobuf/27.1/lib/libprotobuf.27.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/Cellar/protobuf/27.1/lib/libprotobuf.27.0.0.dylib' (no such file), '/usr/local/Cellar/protobuf/27.1/lib/libprotobuf.27.0.0.dylib' (no such file)
[Err] [SystemLoader.cc:107] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [GstCameraPlugin]
- Requested library name: [GstCameraPlugin]
- Resolved library path: [/Users/sergiir/git/gz_ws/src/ardupilot_gazebo/build/libGstCameraPlugin.dylib]```

srizvan, 2024-07-08:
I've upgraded OpenCV to the latest version and everything works! Finally!

```brew install opencv```

srmainwaring, 2024-07-08:
@srizvan it seems that the behaviour for handling legacy materials has changed in a recent update to Gazebo. I’ll need to track down why that is affecting a number of models loaded from fuel containing material scripts on macOS. Not sure if it is also an issue on Ubuntu. This is not an AP specific problem but does affect some of the examples.

@IuliuNovac the problem with gstreamer  you are seeing is a separate issue and unrelated to the material scripts. If it’s still causing a problem please open an issue and provide details of your system etc so we can try and replicate.

IuliuNovac, 2024-07-11:
> @srizvan it seems that the behaviour for handling legacy materials has changed in a recent update to Gazebo. I’ll need to track down why that is affecting a number of models loaded from fuel [...]

Latest updated in main branch fixed the gstreamer issue. Thanks!

The material scripts are affected on ubuntu as well.

closed 2024-10-30 by srmainwaring