ArduPilot/ardupilot_gazebo #109: iris_with_standoffs import error when using Gazebo Harmonic and ROS Jazzy

TannerGilbert, 2024-09-01:
## Bug report

**Issue details**

I'm running [ROS Jazzy](https://docs.ros.org/en/jazzy/index.html) and [Gazebo Harmonic](https://gazebosim.org/docs/latest/install/) and tried installing ardupilot_gazebo following the installation instructions on the ros2 branch. Installation works as expected but when I run `gz sim -v4 -r iris_runway.sdf` I get the following expections:

```
[Err] [Server.cc:204] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [package://ardupilot_gazebo/models/iris_with_standoffs/meshes/iris.dae] could not be resolved.
[Err] [Server.cc:204] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [package://ardupilot_gazebo/models/iris_with_standoffs/meshes/iris_prop_ccw.dae] could not be resolved.
[Err] [Server.cc:204] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [package://ardupilot_gazebo/models/iris_with_standoffs/meshes/iris_prop_ccw.dae] could not be resolved.
[Err] [Server.cc:204] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [package://ardupilot_gazebo/models/iris_with_standoffs/meshes/iris_prop_cw.dae] could not be resolved.
[Err] [Server.cc:204] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [package://ardupilot_gazebo/models/iris_with_standoffs/meshes/iris_prop_cw.dae] could not be resolved.
[Err] [Server.cc:204] Error Code 9: Msg: Failed to load a world.
```

This seems to be related to the changes in #56 as the problem doesn't occur when changing from:
```diff
+ <uri>package://ardupilot_gazebo/models/iris_with_standoffs/meshes/iris.dae</uri>
- <uri>model://iris_with_standoffs/meshes/iris.dae</uri>

+ <uri>model://iris_with_standoffs/meshes/iris_prop_ccw.dae</uri> 
- <uri>package://ardupilot_gazebo/models/iris_with_standoffs/meshes/iris_prop_ccw.dae</uri>

+ <uri>model://iris_with_standoffs/meshes/iris_prop_cw.dae</uri>
- <uri>package://ardupilot_gazebo/models/gimbal_small_2d/meshes/base_main.dae</uri>
```

I don't have extensive experience with Gazebo and therefore I'm unsure if this is causes by incorrect usage or not but I suggest that either the README should be adapted to show the correct usage of the change should be reverted.

**Version**

Current version of ros branch (77b303e)

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
`iris_with_standoffs` (used also in other iris models)

srmainwaring, 2024-09-02:
@TannerGilbert if you are using the `ros2` branch instead of `main` it is expected that the Gazebo session is started using a ros2 launch script and the packages will be resolved using the ros2 pkg system. The reason for the different prefix is that the `sdformat_urdf` plugin for the joint state publisher only works with the package prefix `<uri>package://...</uri>`.

To run using `gz sim etc` in the command line use the `main` branch.

TannerGilbert, 2024-09-02:
Thanks @srmainwaring for the quick response. The information about the `sdformat_urdf` plugin is really helpful as I already wondered why `robot_state_publisher` wouldn't work with a SDF model previously. Could you perhaps add an example on how to correctly start the Gazebo session from a ROS launch script to the README of the ros2 branch?

The following launch file allowed me to correctly spawn the drone but I didn't get the `robot_state_publisher`and `joint_state_publisher` to work correctly:
```python
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import ExecuteProcess

def generate_launch_description():
    sdf_file_path = os.path.join(
        get_package_share_directory("ardupilot_gazebo"), "worlds", "iris_runway.sdf"
    )

    start_gazebo = ExecuteProcess(
        cmd=["gz", "sim", "-v4", "-r", sdf_file_path], output="screen"
    )

    return LaunchDescription([start_gazebo])
```

Many thanks for the help and best regards,
Gilbert

srmainwaring, 2024-09-02:
Hi @TannerGilbert, the documentation for launching Gazebo with ROS is here: https://ardupilot.org/dev/docs/ros2-gazebo.html.

There are launch examples for just the iris, the iris with nav2 enabled, and an example for a skid-steer rover. Setting up the example is more involved, as the objective is to use the onboard ROS2/DDS support in ArduPilot.

TannerGilbert, 2024-09-02:
Thank you for all the help. I'll check this out in more detail as soon as possible.

closed 2024-09-02 by TannerGilbert