ArduPilot/ardupilot_gazebo #111: Fail to display the streamed video in QGC, but gazebo can show the gimbal topic correctly

prodrone-liu, 2024-09-17:
## Bug report

**Issue details**
QGC can not render the video stream according to the [guide](https://github.com/ArduPilot/ardupilot_gazebo?tab=readme-ov-file#3-streaming-camera-video). The camera topic can be seen from the image display in gazebo-garden, but QGC can not display the same video stream with default port 5600. ArduPilot SITL console can be launched successfully and used to control the iris in gazebo. 

Running gst-launch-1.0:
```shell
gst-launch-1.0 -v udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink sync=false
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstUDPSrc:udpsrc0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264
``` 

Running gz topic and find  the msg is not publishing at 127.0.0.1. Is that normal?
```shell
gz topic -i -t /world/iris_runway/model/iris_with_gimbal/model/gimbal/link/pitch_link/sensor/camera/image
Publishers [Address, Message Type]:
  tcp://172.16.6.163:38773, gz.msgs.Image
``` 

QGC can not find the camera:
```shell
./GCS\ Software/QGroundControl.AppImage 
Settings location "/home/xliu/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/xliu/Documents/QGroundControl/Logs"
Map Cache in: "/home/xliu/.cache/QGCMapCache300" / "qgcMapCache.db"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
"v4.3.0"
Adding target QHostAddress("127.0.0.1") 36638
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
FactPanelControllerLog: Missing parameter: "1:MNT_RC_IN_PAN"
FactPanelControllerLog: Missing parameter: "1:MNT_STAB_PAN"
FactPanelControllerLog: Missing parameter: "1:MNT_ANGMIN_PAN"
FactPanelControllerLog: Missing parameter: "1:MNT_ANGMAX_PAN"
FactPanelControllerLog: Missing parameter: "1:MNT_RC_IN_ROLL"
FactPanelControllerLog: Missing parameter: "1:MNT_STAB_ROLL"
FactPanelControllerLog: Missing parameter: "1:MNT_ANGMIN_ROL"
FactPanelControllerLog: Missing parameter: "1:MNT_ANGMAX_ROL"
FactPanelControllerLog: Missing parameter: "1:MNT_RC_IN_TILT"
FactPanelControllerLog: Missing parameter: "1:MNT_STAB_TILT"
FactPanelControllerLog: Missing parameter: "1:MNT_ANGMIN_TIL"
FactPanelControllerLog: Missing parameter: "1:MNT_ANGMAX_TIL"
FactPanelControllerLog: Missing parameter: "1:MNT_RETRACT_X"
FactPanelControllerLog: Missing parameter: "1:MNT_RETRACT_Y"
FactPanelControllerLog: Missing parameter: "1:MNT_RETRACT_Z"
FactPanelControllerLog: Missing parameter: "1:MNT_NEUTRAL_X"
FactPanelControllerLog: Missing parameter: "1:MNT_NEUTRAL_Y"
FactPanelControllerLog: Missing parameter: "1:MNT_NEUTRAL_Z"
FactPanelControllerLog: Missing parameter: "1:BRD_PWM_COUNT"
qrc:/qml/APMCameraComponent.qml:193: TypeError: Cannot read property 'value' of null
VideoReceiverLog: Stop called on empty URI
Quit event true
Done
VideoReceiverLog: Stop called on empty URI
After app delete
``` 

The gimbal camera is actually working in gazebo-garden:
[image]

**Version**
Master branch

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

prodrone-liu, 2024-09-18:
Thanks for the help!
I found the reason of failure is due to the configuration of model.sdf file in gimbal_small_3d. The tag <use_cuda> is set to true so the GstCameraPlugin failed when I ran the streaming enable command:
```shell
gz topic -t /world/iris_runway/model/iris_with_gimbal/model/gimbal/link/pitch_link/sensor/camera/image/enable_streaming -m gz.msgs.Boolean -p "data: 1"
[Msg] GstCameraPlugin:: streaming: started
[Dbg] [GstCameraPlugin.cc:407] GstCameraPlugin: creating generic pipeline
[Dbg] [GstCameraPlugin.cc:477] Using Cuda
(gz sim server:8009): GLib-GObject-CRITICAL **: 11:26:59.773: g_object_set: assertion 'G_IS_OBJECT (object)' failed
[Err] [GstCameraPlugin.cc:419] GstCameraPlugin: failed to create GStreamer elements
[Msg] GstCameraPlugin: starting GStreamer main loop
``` 
If I set it back to false, everything is normal and I can get the video stream both in gst-launch-1.0 and QGC. Not sure why cuda is not working...
Driver info:
```shell
prime-select query
nvidia

nvidia-smi
Wed Sep 18 11:29:14 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4060 ...    Off | 00000000:01:00.0  On |                  N/A |
| N/A   48C    P4              12W /  55W |   1994MiB /  8188MiB |     39%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

```

srmainwaring, 2024-09-18:
@prodrone-liu, unfortunately I do not have a machine with an Nvidia card to test the `<use_cuda>` option. It is set to false by default in the examples distributed with the plugins.