ArduPilot/ardupilot_gazebo #130: Zephyr (fixed wing plane) sim doesn't work on MacOS Sequoia

Tensho, 2025-03-12:
## Bug report

**Issue details**

https://discuss.ardupilot.org/t/ardupilot-gazebo-zephyr-arduplane-sitl-model-doesnt-work-on-macos-sequoia/131581

Run:
```
$ sim_vehicle.py -v ArduPlane --add-param-file=$HOME/Projects/SITL_Models/Gazebo/config/skywalker_x8.param --model JSON --map --console --wipe-eeprom
...
EOF on TCP socket
Attempting reconnect
[Errno 61] Connection refused sleeping
```
SITL log:
```
/Users/tensho/Projects/ardupilot/build/sitl/bin/arduplane -w --model JSON --speedup 1 --slave 0 --defaults ../../../../ardupilot/Tools/autotest/models/plane.parm,/Users/tensho/Projects/SITL_Models/Gazebo/config/skywalker_x8.param --sim-address=127.0.0.1 -I0
Setting SIM_SPEEDUP=1.000000
Starting SITL: JSON
JSON control interface set to 127.0.0.1:9002
Starting sketch 'ArduPlane'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for SERIAL0
SERIAL0 on TCP port 5760
Waiting for connection ....
Connection on serial port 5760
Loaded defaults from ../../../../ardupilot/Tools/autotest/models/plane.parm,/Users/tensho/Projects/SITL_Models/Gazebo/config/skywalker_x8.param
Home: -35.363262 149.165237 alt=584.000000m hdg=353.000000

JSON received:
	timestamp
	imu: gyro
	imu: accel_body
	position
	quaternion
	velocity

bind port 5762 for SERIAL1
SERIAL1 on TCP port 5762
bind port 5763 for SERIAL2
SERIAL2 on TCP port 5763
validate_structures:526: Validating structures
Loaded defaults from ../../../../ardupilot/Tools/autotest/models/plane.parm,/Users/tensho/Projects/SITL_Models/Gazebo/config/skywalker_x8.param
[1]    35630 illegal hardware instruction  /Users/tensho/Projects/ardupilot/build/sitl/bin/arduplane -w --model JSON  1
```

**Version**

MacOS: 15.3.1 (Sequoia)
Gazebo: Garden, Harmonic, Ionic

All dependencies have been installed with Homebrew:

```
git clone https://github.com/ArduPilot/ardupilot_gazebo
cd ardupilot_gazebo
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_PREFIX_PATH="/opt/homebrew/opt/qt@5"
make -j4
```

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

[Zephyr](https://github.com/ArduPilot/ardupilot_gazebo/tree/main/models/zephyr_with_ardupilot), [Skywalker X8](https://github.com/ArduPilot/SITL_Models/tree/master/Gazebo/models/skywalker_x8)

**Hardware type**

No HITL, only SITL. ArduPilot lastest `master` branch ([aa43a56](https://github.com/ardupilot/ardupilot/commit/aa43a5688c8279fc46754382e5eca27cc07d65e3)).

**Logs**

https://discuss.ardupilot.org/t/ardupilot-gazebo-zephyr-arduplane-sitl-model-doesnt-work-on-macos-sequoia/131581/2?u=tensho

srmainwaring, 2025-03-12:
@Tensho, this may not be Gazebo specific. If you run a standard SITL plane simulation (no Gazebo), do you still see the hardware instruction error? The issue should not be present in the debug build (use the `--debug` flag). If it's a SITL specific error we should move the issue to the main ArduPilot repo.

Tensho, 2025-03-12:
The standalone ArduPilot SITL works fine:
```
$ sim_vehicle.py -v ArduPlane --add-param-file=$HOME/Projects/SITL_Models/Gazebo/config/skywalker_x8.param --debug --console --map
...
SIM_VEHICLE: Using defaults from (../../../../ardupilot/Tools/autotest/models/plane.parm)
SIM_VEHICLE: Adding parameters from (/Users/tensho/Projects/SITL_Models/Gazebo/config/skywalker_x8.param)
SIM_VEHICLE: Run ArduPlane
SIM_VEHICLE: "/Users/tensho/Projects/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduPlane" "/Users/tensho/Projects/ardupilot/build/sitl/bin/arduplane" "--model" "plane" "--speedup" "1" "--slave" "0" "--defaults" "../../../../ardupilot/Tools/autotest/models/plane.parm,/Users/tensho/Projects/SITL_Models/Gazebo/config/skywalker_x8.param" "--sim-address=127.0.0.1" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--retries" "5" "--out" "127.0.0.1:14550" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--map" "--console"
Connect tcp:127.0.0.1:5760 source_system=255
[Errno 61] Connection refused sleeping
[Errno 61] Connection refused sleeping
Loaded module console
2025-03-12 11:22:48.330 Python[55047:7661409] +[IMKClient subclass]: chose IMKClient_Modern
2025-03-12 11:22:48.982 Python[55048:7661435] +[IMKClient subclass]: chose IMKClient_Modern
Loaded module map
Running script (/Users/tensho/.mavinit.scr)
Loaded module graph
Loaded module msg
Loaded module devop
Loaded module message
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Detected vehicle 1:1 on link 0
INITIALISING> MANUAL> Received 1424 parameters (ftp)
Saved 1424 parameters to mav.parm

MANUAL> arm throttle
MANUAL> mode takeoff
```
Also, when I run standalone Gazebo world + model everything is fine too:
```
gz sim -v 4 -r zephyr_runway.sdf -s
```
```
gz sim -v 4 -g
```
[image]

The issue is exactly with the bridge – ArduPilot Gazebo plugin.

Tensho, 2025-03-12:
The standalone ArduPilot SITL works fine:
```
$ sim_vehicle.py -v ArduPlane --debug --console --map
...
SIM_VEHICLE: Using defaults from (../../../../ardupilot/Tools/autotest/models/plane.parm)
SIM_VEHICLE: Run ArduPlane
SIM_VEHICLE: "/Users/tensho/Projects/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduPlane" "/Users/tensho/Projects/ardupilot/build/sitl/bin/arduplane" "--model" "plane" "--speedup" "1" "--slave" "0" "--defaults" "../../../../ardupilot/Tools/autotest/models/plane.parm" "--sim-address=127.0.0.1" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--retries" "5" "--out" "127.0.0.1:14550" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--map" "--console"
Connect tcp:127.0.0.1:5760 source_system=255
[Errno 61] Connection refused sleeping
Loaded module console
2025-03-12 11:32:45.372 Python[60990:7696475] +[IMKClient subclass]: chose IMKClient_Modern
2025-03-12 11:32:45.659 Python[60991:7696479] +[IMKClient subclass]: chose IMKClient_Modern
Loaded module map
Running script (/Users/tensho/.mavinit.scr)
Loaded module graph
Loaded module msg
Loaded module devop
Loaded module message
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Detected vehicle 1:1 on link 0
INITIALISING> MANUAL> Received 1434 parameters (ftp)
Saved 1434 parameters to mav.parm

MANUAL> arm throttle
MANUAL> mode takeoff
```
Also, when I run standalone Gazebo world + model everything is fine too:
```
gz sim -v 4 -r zephyr_runway.sdf -s
```
```
gz sim -v 4 -g
```
[image]

The issue is exactly with the bridge – ArduPilot Gazebo plugin.

srmainwaring, 2025-03-12:
> `$ sim_vehicle.py -v ArduPlane --debug --console --map`

I'd expect this to work in `debug`. Does the non-debug version work as well?

> `$ sim_vehicle.py -v ArduPlane --console --map`

The error is not in the plugin or Gazebo, as the `35630 illegal hardware instruction  /Users/tensho/Projects/ardupilot/build/sitl/bin/arduplane` indicates it's in the AP binary.

I've seen random, very hard to reproduce floating point errors in AP from time to time when compiling in release mode with Apple clang compilers. It's never been a Gazebo issue, and it's very hard to track them in AP, since as they don't occur in debug builds and small code ordering changes such as adding print statements near the problem hide the issue.

The work around on macOS is to use the `--debug` flag in the SITL command line.

Tensho, 2025-03-12:
@srmainwaring I understand it could be difficult to track down this type of issues. When I enabled debug mode (`--debug`) and `sim_vehicle.py` script recompiled ArduPlane, there was no extra input appeared 🤷‍♂ 

Totally randomly, I found everything works as expected if I run ArduPlane SITL **first** and Zephyr Gazebo world + model **second** – no illegal hardware instruction error.
1. Run ArduPlane SITL with JSON model:
```shell
$ sim_vehicle.py -v ArduPlane -f gazebo-zephyr --debug --console --map --model JSON
```
2. Run Gazebo server:
```shell
$ gz sim -v 4 -r zephyr_runway.sdf -s
```
3. Run Gazebo GUI:
```shell
$ gz sim -v 4 -g
```
Nonetheless, this order doesn't make Skywalker X8 model work 🤔 Closing this ticket in favour of another in [SITL_Models](https://github.com/ArduPilot/SITL_Models) repository.