ArduPilot/ardupilot_gazebo #139: IMU orientation not reflected in TF tree

TannerGilbert, 2025-05-19:
## Bug report

**Issue details**

For the Iris with Standoff model the IMU is rotated 180° around the X axis (https://github.com/ArduPilot/ardupilot_gazebo/blob/main/models/iris_with_standoffs/model.sdf#L157) but this is not reflected in the TF tree between `base_link` and `imu_link` leading to confusion when relying on the TF tree for extrinsics. 

E.g.: `ros2 run tf2_ros tf2_echo base_link imu_link`

Output:
```
At time 6.800000000
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [0.000, -0.000, 0.000]
- Rotation: in RPY (degree) [0.000, -0.000, 0.000]
- Matrix:
  1.000 -0.000  0.000  0.000
  0.000  1.000  0.000  0.000
  0.000  0.000  1.000  0.000
  0.000  0.000  0.000  1.000
```

**Version**
Tested on Ubuntu 22.04, ROS 2 Humble and Gazebo Harmonic following https://ardupilot.org/dev/docs/ros2-gazebo.html

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Iris

**Hardware type**
- (Simulation)

srmainwaring, 2025-05-19:
@TannerGilbert , the rotation is in the sensor pose rather than the IMU link, so I would not expect the tf from base_link to imu_link to include the rotation. It should appear in the IMU frame (if we have added the frame_id element to the model).

TannerGilbert, 2025-05-19:
@srmainwaring thanks for the fast reply. Unfortunatelly I don't fully get what you mean with the frame_id of the IMU as currently the IMU (in the ROS 2 branch) has a `gz_frame_id` equal to `imu_link`. Would it need a unqiue frame_id?

Current sensor setup:
```
<sensor name="imu_sensor" type="imu">
  <gz_frame_id>imu_link</gz_frame_id>
  <pose degrees="true">0 0 0 180 0 0</pose>
  <always_on>1</always_on>
  <update_rate>1000.0</update_rate>
</sensor>
```

srmainwaring, 2025-05-19:
Ah, yes that could be the issue. 

A few options,

1. move the IMU sensor to the base link and retain the additional gz_frame_id

2. Move the rotation to the imu_link and do not rotate the sensor (may have an impact in the ArduPilot plugin transform element?)

3. Rename the gz_frame_id

The first is probably the simplest and gets rid of a fixed joint (good).

gz_frame_id is deprecated in favour of frame_id in Ionic, so there is some renaming to do eventually.

TannerGilbert, 2025-05-19:
Thanks for the quick reply. I tested out the described options:
1. Only having `imu_link` as `gz_frame_id` doesn't put it into the TF tree
2. IMU output and TF tree look good but when trying to takeoff I receive `PreArm: Check mag field (z diff: 1055>200)
3. Similar to 1 doesn't create a entry into the TF tree for me